smartdevicelink/sdl_core #3124: [SDL 0256] Refactor Fuel Information Related Vehicle Data

theresalech, 2019-11-14:
Proposal: [Refactor Fuel Information Related Vehicle Data](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0256-Refactor-Fuel-Information-Related-Vehicle-Data.md)

> This proposal is to consolidate Fuel Information related vehicle data so that it supports multiple types of propulsion sources used by the vehicle.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/854

Steering Committee Decision:

>The Steering Committee voted to accept this proposal with revisions.  The author will make the following changes to the proposal: add `KILOGRAM` to `CapacityUnit`; change maxValue of `level` and `capacity` in `FuelRange` to larger upper limit (1000000); fix `fuelLevel` and `fuelLevel_State` typos; update descriptions of `fuelLevel` and `fuelLevel_State` to state the param is deprecated as of x.x.x of the rpc spec, please see `fuelRange`; modify description of `fuelRange` param to be more descriptive of its extended capabilities.  The author can work with the Project Maintainer to finalize this description.

> The proposal .md file was updated to reflect these revisions on 11/14/19.

theresalech, 2020-03-27:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0256 Revisions - Refactor Fuel Information Related Vehicle Data](smartdevicelink/sdl_evolution#972). The revisions include updating missing type for structure and adding history for deprecated parameters in Mobile_API.  Please reference the issue and proposal markdown file for more details.

ShobhitAd, 2020-08-06:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3282

closed 2020-08-06 by ShobhitAd