smartdevicelink/sdl_core #3133: [SDL 0255] Enhance BodyInformation vehicle data

jordynmackool, 2019-11-22:
Proposal: [Enhance BodyInformation vehicle data](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0255-Enhance-BodyInformation-vehicle-data.md)

> This proposal is to enhance BodyInformation with more params to get access to more door related signals.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/847#issuecomment-555798045

theresalech, 2020-06-09:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0255 Revisions - Enhance BodyInformation vehicle data](https://github.com/smartdevicelink/sdl_evolution/issues/1022).  Please reference the issue and proposal markdown file for revisions details.

iCollin, 2021-01-26:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3584

closed 2021-01-26 by iCollin