smartdevicelink/sdl_core #3140: SDL keep sending OnVideoDataStreming notification to HMI

mkorniichuk, 2019-11-25:
### Bug Report
SDL sends extra OnVideoDataStreaming notification in case Projection application stream is suspended

##### Reproduction Steps
1. Register and activate Projection application
2. Start streaming service but don't start streaming data transmission

##### Expected Behavior
SDL Suspends application and sends 
`Navigation.OnVideoDataStreaming {"available" = false}`
 to HMI once

##### Observed Behavior
Application is suspended, but notifications keep spamming to HMI.

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: develop (891497bc9fb23434242dbe92e1f13dd1273fcb21)
* Testing Against: ATF

dboltovskyi, 2019-12-06:
I seems issue is related to pure connection.
In case if Video packet is not received by SDL during 1 sec (defined by `VideoDataStoppedTimeout` .ini parameter) SDL sends `Navigation.OnVideoDataStreaming {"available" = false}`
Once new packet is received SDL sends `Navigation.OnVideoDataStreaming {"available" = true}`

dboltovskyi, 2019-12-23:
Description is corrected, issue is reopened

dboltovskyi, 2020-01-08:
@theresalech Please triage this issue

theresalech, 2020-01-08:
@dboltovskyi this issue has not yet been raised to the Steering Committee as priority for the upcoming Core release.  We can work to include if time allows, but would recommend raising as priority in the Steering Committee meeting if it's imperative the issue be included.  Thanks!

iCollin, 2020-01-29:
Closed via #3202

PR #3202 (2020-01-29): Stop app unregister timer after streaming session restarted
Fixes #3140

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
SDL starts a timer for an app to be unregistered if it continues streaming. And when timeout expired and Application already stopped service - nothing would happen. But if the application restarts streaming service, SDL would not know that it is a new session. And so, in this case, after timer expiration, SDL will unregister an application, despite the fact it is a brand new streaming session. Current PR implements the cleaning up of [... cut]
closed 2020-01-29 by iCollin