smartdevicelink/sdl_core #3141: Needed changes in command request inheritance structure

YarikMamykin, 2019-11-25:
### Bug Report
SDL Crash observed on invocation on_event method of commands.

##### Reproduction Steps
Execute ING_OFF/IGN_ON.
Trigger some simple command, e.g. VIIsReadyRequest.

##### Observed Behavior
SDL core crash random, not often occurences.

Occurrence: random

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a top priority for Toyota. Does this occur with the reproduction steps described? Also, does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@YarikMamykin, can you please advise if the Luxoft team is seeing this issue in Core 6.1?

LuxoftAKutsan, 2020-05-19:
@theresalech  sdl_core 6.1 do not contains appropriate code. 
Issue reproduced on SYNC environment. 
Script for open sdl will be created. But issue can be not reproducible on open environment. 
SYNC fixes requires major reworking to be compatible with 6.1

theresalech, 2020-05-19:
@LuxoftAKutsan, thank you for the update.  Please let us know once the script is created, and if a PR can be submitted to the open for SYNC alignment.