smartdevicelink/sdl_core #3144: SDL doesn't increment 'count_of_rejections_duplicate_name' counter

dboltovskyi, 2019-11-25:
### Bug Report
SDL doesn't increment 'count_of_rejections_duplicate_name' counter

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile
2. Register App_1 with name, e.g. `Test App`
3. Register App_2 with another name
4. Unregister App_2
5. Register App_2 one more time with name `Test App`

##### Expected Behavior
SDL rejects App_2 with `DUPLICATE_NAME` result code and increment `count_of_rejections_duplicate_name` counter

##### Observed Behavior
SDL rejects App_2 with `DUPLICATE_NAME` result code and does not increment counter

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (e137556)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2295
 - Report & logs: [2019-11-25_14-29-16.tar.gz](https://github.com/smartdevicelink/sdl_core/files/3888357/2019-11-25_14-29-16.tar.gz)

dboltovskyi, 2019-11-25:
@theresalech Please triage this issue

jordynmackool, 2019-11-26:
@dboltovskyi, are you planning on submitting a PR for this issue?

dboltovskyi, 2019-11-27:
@jordynmackool Yes, we're planning to open PR with fix

jordynmackool, 2019-11-27:
@dboltovskyi please inform us when the PR is ready for Livio review.

dboltovskyi, 2019-12-03:
@theresalech Please notice all required updates are ready for PM review:
 - core: 
https://github.com/smartdevicelink/sdl_core/pull/3147
 - scripts:
  https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2295
  https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2299

iCollin, 2020-01-08:
Closed via #3147

PR #3147 (2020-01-08): Fix increment of 'count_of_rejections_duplicate_name' counter
Fixes #3144 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF test scripts attached to #3144 

### Summary
In case of registering app with a duplicated name SDL doesn't increase counter of duplicated names. Added incrementing counter of rejected apps with duplicated names.

### CLA
- [X] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-01-08 by iCollin