smartdevicelink/sdl_core #3148: Deadlock happens during video streaming

zhouxin627, 2019-11-29:
### Bug Report
Deadlock happens during video streaming

##### Preconditions
Set `VideoDataStoppedTimeout = 500` in smartDeviceLink.ini

The smaller the value of VideoDataStoppedTimeout is, the more likely the deadlock happens.
If `VideoDataStoppedTimeout `is 3000, it is almost impossible to reproduce the issue.

##### Reproduction Steps
1. Register Sygic
2. Activate Sygic, starts video streaming
3. Quickly and repeatedly click the screen of Sygic on the head unit

##### Expected Behavior
Deadlock should not happen.

##### Observed Behavior
Deadlock happens during video streaming.

The reason seems to be that WakeUpStreaming happens deadlock.
If the phone does not sends streaming data, a timeout occurs.
After SuspendStreaming, WakeUpStreaming is started, then a deadlock happens. 
Please refer to the attached logs for further details.

##### OS & Version Information
* Sygic Version:18.0.0
* Android Version: Android 9
* SDL Android Version: 4.9.0
* Testing Against: SDL Core 5.1.3 + HMI running on our development environment.

##### Test Case, Sample Code, and / or Example App
Logs:logs.txt
[logs.txt](https://github.com/smartdevicelink/sdl_core/files/3903549/logs.txt)

[Paste a link to a PR, gist, or other code that exemplifies this behavior]

E-SAITO-TMC, 2019-11-29:
This is one of the issues that Toyota reported at the 11/18 Roadmap workshop.

theresalech, 2019-12-02:
@E-SAITO-TMC is the Toyota team working on a fix for this issue? If so, can you please let us know when the fix will be ready for Livio review?

E-SAITO-TMC, 2019-12-03:
@theresalech 
I'm sorry, I found a bug.
It has not been fixed yet.
We will let you know when a solution is found.

iCollin, 2020-02-10:
Closed via #3231

PR #3231 (2020-02-10): Fix deadlock happens during video streaming
Fixes #3148 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by Unit tests.

### Summary
**Reason**
A deadlock occurs between `WakeUpStreaming` and `SuspendStreaming`.
If the Video App does not send streaming data, a timeout occurs.
After `SuspendStreaming`, `WakeUpStreaming` is started, then the deadlock occurs.

Audio streaming has the same problem，a deadlock may occur between `WakeUpStreaming` and `SuspendStreaming` because of `audio_streaming_suspended_lock_` Lock.

**Solution**
`video_strea [... cut]
closed 2020-02-10 by iCollin