smartdevicelink/sdl_core #3149: SDL Core crash after quickly and repeatedly connect/disconnect USB cable

zhouxin627, 2019-11-29:
##### Bug Report
SDL Core crash after quickly and repeatedly connect/disconnect the USB cable

##### Preconditions
Install more than 3 SDL apps on the phone

##### Reproduction Steps
Quickly and repeatedly connect/disconnect USB cable

##### Expected Behavior
SDL core should not crash

##### Observed Behavior
SDL Core crash after quickly and repeatedly connect/disconnect the USB cable.

The reason seems to be that AM Pool is not thread-safe, it is possible to use the Object just removed when users quickly and repeatedly connect/disconnect the USB cable. 
Please refer to the attached logs for further details.

`src/components/application_manager/src/policies/policy_handler.cc`

##### OS & Version Information
* iOS Version:10.2
* SDL iOS Version:6.3.0
* Android Version: Android 9
* SDL Android Version: 4.9.0
* Testing Against: SDL Core 5.1.3 + HMI running on our development environment.

##### Test Case, Sample Code, and / or Example App
Logs:logs.txt
[logs.txt](https://github.com/smartdevicelink/sdl_core/files/3903624/logs.txt)

E-SAITO-TMC, 2019-11-29:
This is one of the issues that Toyota reported at the 11/18 Roadmap workshop.

theresalech, 2019-12-02:
@E-SAITO-TMC is the Toyota team working on a fix for this issue? If so, can you please let us know when the fix will be ready for Livio review?

E-SAITO-TMC, 2019-12-03:
@theresalech 
I'm sorry, I found a bug.
It has not been fixed yet.
We will let you know when a solution is found.

JackLivio, 2020-02-10:
resolved by #3230

PR #3230 (2020-02-10): Fix ApplicationSet reference variables are not lock-protected
Fixes #3149 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by Unit tests.

### Summary
**Reason**
When App registered and another App unregistered, multiple threads access the same variable: application.
`AM FromMobile` thread free `applicationPtr`, and `AM Pool` thread uses the same `applicationPtr` which had freed by `AM FromMobile` thread.

**Solution**
`ApplicationSet` reference variables are not lock-protected, change it to a local variable.

**Sequence**
[image]

### CL [... cut]
closed 2020-02-10 by JackLivio