smartdevicelink/sdl_core #3156: CCB: Fixed core crash on accesing the application pointer

LitvinenkoIra, 2019-12-05:
### Bug Report
iOS switch from BT to USB is causing SDL to crash

### Detailed analysis:
App pointer check missing from ApplyPostponedStateForApp() function.

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile
2. iOS initially paired to B/T
3. Launch SYNC Proxy Tester
4. Go to Apps Page
5. Connect Same iOS device via USB
6. Observe behavior

##### Expected Behavior
SDL should not crash

##### Observed Behavior
When Apps un-register and re-register over USB, SDL is crashing

#### Occurrence: 
Random

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

LitvinenkoIra, 2019-12-10:
[sdl crash 2.zip](https://github.com/smartdevicelink/sdl_core/files/3945736/sdl.crash.2.zip)

JackLivio, 2020-02-12:
Fixed with merge of #3194

PR #3194 (2020-02-12): Fix core crash on accesing the application pointer.
Fixes #3156

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
The issue will occur when the state controller gets an application collection from the application manager and tries to apply a functor to each application.
One of the application's pointers could be uninitialized and the state controller should check if pointer is valid.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-02-12 by JackLivio