smartdevicelink/sdl_core #3160: CCB: Fix streamer activity in case of suspend

LitvinenkoIra, 2019-12-05:
### Bug Report
SDL stops streaming data by timeout when no new data from mobile received

### Detailed analysis:
The problem of the current implementation is that SDL kills streaming thread responsible for sending a/v streaming data to HMI when streaming timeout expires.
This issue is observed when mobile app dumps 10-seconds audio file to SDL during 2 seconds and after that app does not send any data. In that case, HMI will play audio file during 2 seconds + timeout = 5 seconds. At the 5th second SDL just kills streaming thread with all pending messages, however, audio service is still open. As a result, not a whole audio file is played.

The correct behavior from SDL side in that case is not kill streaming thread when streaming timeout was expired. SDL should kill streaming thread only when service is
actually stopped.

##### Reproduction Steps
1.Connect an iOS device to SYNC via USB
2.Launch Nav App on Mobile device and then on SYNC
3. Go to Menu options on the mobile device --> AV Streaming -->Under Audio Streaming Encrypt Video to NONE --> Start session --> Start Audio
4. Wait until the whole audio file is played

##### Expected Behavior
Should be played whole audio file

##### Observed Behavior
Audio is played just a few seconds

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

AKalinich-Luxoft, 2020-09-04:
@theresalech fix has been submitted and ready for Livio review

iCollin, 2020-10-01:
Issue as described was fixed by https://github.com/smartdevicelink/sdl_core/pull/3057. 

Issue title and PR #3488 seem to suggest there is a problem when a stream is suspended, can you please update the reproduction steps and/or description to reflect that?

AKalinich-Luxoft, 2020-10-02:
> Issue as described was fixed by #3057.
> 
> Issue title and PR #3488 seem to suggest there is a problem when a stream is suspended, can you please update the reproduction steps and/or description to reflect that?

PR #3057 is just solves audio playback issue on socket transport. PR #3488 suggest another solution for all the transports

iCollin, 2020-10-05:
Closed via #3488

PR #3488 (2020-10-05): Fix streamer activity in case of suspend
Fixes #3160 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
[Describe how you plan to unit test the changes in this PR]

### Summary
The problem of the current implementation is that SDL kills streaming thread responsible for sending a/v streaming data to HMI when streaming timeout expires. This issue is observed when mobile app dumps 10-seconds audio file to SDL during 2 seconds and after that app does not send any data. In that case HMI will play audio file during 2 seconds + timeout = 5 seconds. At the [... cut]
closed 2020-10-05 by iCollin