smartdevicelink/sdl_core #3162: CCB: SDL Crashes while testing QT 5.12

LitvinenkoIra, 2019-12-06:
### Bug Report
SDL Crashes while testing QT 5.12

### Detailed analysis:
SDL added the new application registered to the applicationlist but failed to update the plugin manager before "onVehicleData" notification is received from the HMI from different thread, since the application list contains extra application than to the registered app extension, it was throwing null extension ptr which while referencing caused a crash by the RPC handler thread. Worst case the delay between the application list update and extension update was about 70-100ms and before that if onvehicledata is received,then it will result in crash.

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile
2. Connect iOS with SYNC via BT. Disconnect BT reconnect BT  -->  SDL Crash #1 observed
3. Connect iOS with SYNC via USB + Android via BT – Unplug iOS --> SDL Crash observed

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashed

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a high priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

theresalech, 2020-05-21:
@LitvinenkoIra can you please advise if you are able to reproduce this issue in the latest Core release?

LuxoftAKutsan, 2020-05-27:
Fix from SYNC 4 is partially applicable to open sdl. Will be applied. 
Only regression will be checked because issue related to IOs devices

theresalech, 2020-05-27:
@LuxoftAKutsan, thank you for the additional information!  Please let us know when a PR is ready for Livio's review.

closed 2020-10-06 by JackLivio