smartdevicelink/sdl_core #3167: CCB: Projection/Navigation app not going to FULL

LitvinenkoIra, 2019-12-06:
### Bug Report
Projection/Navigation app is not getting HMI Status when "Enable Mobile Apps" option is disabled

### Detailed analysis:
Deadlock causing the iAP2 protocol threads to increase without control due to missing acknowledge from SDL to a StartServiceACK from Mobile.

##### Reproduction Steps
1. Toggle off "Enable Mobile Apps"
2. Register SPT with type Projection/Navigation
3. Launch
4. A pop up seeking permission should be shown

##### Expected Behavior
App should receive HMIStatus and should launch successfully

##### Observed Behavior
App is stuck in "Waiting for Device" pop up and app did not receive HMIStatus

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

jacobkeeler, 2020-09-04:
@AKalinich-Luxoft Are you able to verify if this was resolved by #3476?

IGapchuk, 2020-09-18:
Hi, @jacobkeeler We have verified the issue on the last develop (1f52a0ef360475ba268c49e1eec8b0a93050f132). The issue wasn't reproduced.
[SmartDeviceLinkCore.log.tar.gz](https://github.com/smartdevicelink/sdl_core/files/5245260/SmartDeviceLinkCore.log.tar.gz)

closed 2020-09-18 by jacobkeeler