smartdevicelink/sdl_core #3168: CCB: Fixes race condition in messagemeter.h

LitvinenkoIra, 2019-12-06:
### Bug Report
iOS/Android USB- Apps are not Unregistering

### Detailed analysis:
Unprotected critical section accessed by two different threads,
timing_map is being modified by clearing the map identifiers causing a
crash in boost::date_time

##### Reproduction Steps
1. Connect an iOS device with Applink compatible Apps downloaded on to the phone to SYNC via USB
2. Verify the Apps Registration on SYNC
3. Unplug the device from SYNC

##### Expected Behavior
Apps should Unregister

##### Observed Behavior
Apps are not Unregistering

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a top priority for Toyota. Does this occur with the reproduction steps described? Also, does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@LitvinenkoIra, can you please advise if the Luxoft team is seeing this issue in Core 6.1?

LuxoftAKutsan, 2020-05-19:
@theresalech
Issue reproduced on SYNC version of SDL.
Issue is related to  iOS device and can't be reproduced in open sdl environment. 
Fix from SYNC ported to open : #3377

jacobkeeler, 2020-06-22:
Closing with the merge of #3377

PR #3377 (2020-06-22): Fixes race condition in messagemeter.h
Fixes #3168

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Tested on SYNC. 
Test not applicable for open sdl

### Summary
Unprotected critical section accessed by two different threads,
timing_map was being modified by clearing the map identifiers causing a
crash in boost::date_time
Review: Change RWLock to Lock, Added FrequencyImpl private method

- [ ] Internal Luxoft review

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/vi [... cut]
closed 2020-06-22 by jacobkeeler