smartdevicelink/sdl_core #3169: CCB: Fix SIGKILL due to not clearing the requestinfo set

LitvinenkoIra, 2019-12-06:
### Bug Report
SignalKill in SDL

### Detailed analysis:
time_sorted_pending_requests_ set containing the RequestInfoPtr is getting released twice.

##### Reproduction Steps
1. Register CitySeeker App.
2. Launch the App.
3. Do a proper Ignition Off / Ignition On.
4. Try the above steps more than one time if it is not reproducible.

##### Expected Behavior
SDL crashed
[bt.txt](https://github.com/smartdevicelink/sdl_core/files/3932159/bt.txt)

##### Observed Behavior
No crash

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a top priority for Toyota. Does this occur with the reproduction steps described? Also, does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@LitvinenkoIra, can you please advise if the Luxoft team is seeing this issue in Core 6.1?

LuxoftAKutsan, 2020-05-19:
@theresalech
Issue reproduced on SYNC version of SDL.
On open source version of SDL issue may be non reproducible.
Script will be created.
Fix from SYNC adopted for 6.1 codebase :  https://github.com/smartdevicelink/sdl_core/pull/3376

theresalech, 2020-05-19:
@LuxoftAKutsan, thank you! We will wait for confirmation if the issue is reproducible in the open source based on the findings of your script.

dboltovskyi, 2020-07-07:
@theresalech Please notice the following findings:

Script for provided use case has been created: [3169.lua.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4885400/3169.lua.tar.gz)
Issue was not reproduced on latest SDL develop - neither with fix nor without it.

There could be several reasons why issue is not reproduced in open source environment :
- Timings on Mobile\HMI communication may be different on CI/developer workstation and real hardware
- QNX platform specific issues may trigger this case
- Different CPU resources available on HU, and developer/CI.

Anyway script can be used to check use case on real hardware with Remote ATF feature.
Nevertheless we think proposed fix should be included since it solves 'potential' problem.

theresalech, 2020-07-08:
@dboltovskyi, thank you for the additional information.  We'll await Luxoft's confirmation of PR #3376 being ready for Livio review.

jacobkeeler, 2020-08-31:
Closing with the merge of #3376

PR #3376 (2020-08-31): Fix SIGKILL due to not clearing the requestinfo set.
Fixes #3169

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Tested on SYNC SDL. 
For open sdl will be created test script

### Summary
Adopted fix from SYNC sdl

### Tasks Remaining:
- [ ] Luxoft internal review 

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-08-31 by jacobkeeler