smartdevicelink/sdl_core #3170: CCB: Fix dangerous work with app commands

LitvinenkoIra, 2019-12-06:
### Bug Report
Crash in AttachSchema SDL

### Detailed analysis:
There was related to a thread safety violation while working with application internal commands
array. This array is a thread-safe itself as it is protected with mutexes, however FindCommand()
function is returning a raw pointer to an internal array element so the external component is able to get access to this element at any time even without locking the mutex. Core crash which was found in DeleteCommand request instance is happening by exactly the same reason -
DeleteCommand is using a raw pointer to find the command, however, this command might be destroyed from another thread at this point in time. As a result, working thread can access to a destroyed object by pointer. As a quick solution, each FindCommand() can be followed
by a command accessor which prevents unexpected command destruction. Working thread will use the temporary copy of command, but not command itself. However, the better solution would be to change the design and to avoid using of the raw pointers.

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile
2. Register app
3. Add some commands and then send DeleteCommand RPC
4. Observe behavior

##### Expected Behavior
SDL crashed
[bt.txt](https://github.com/smartdevicelink/sdl_core/files/3932171/bt.txt)

##### Observed Behavior
No crash

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a top priority for Toyota. Does this occur with the reproduction steps described? Also, does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@LitvinenkoIra, can you please advise if the Luxoft team is seeing this issue in Core 6.1?

LuxoftAKutsan, 2020-05-19:
@theresalech  
Issue reproduced on SYNC version of SDL.
On open source version of SDL issue may be non reproducible.
Script will be created.
Fix from SYNC ported to open :  https://github.com/smartdevicelink/sdl_core/pull/3375

theresalech, 2020-05-19:
@LuxoftAKutsan, thank you! We will wait for confirmation if the issue is reproducible in the open source based on the findings of your script.

dboltovskyi, 2020-07-07:
@theresalech Please notice the following findings:

Script for provided use case has been created: [3170.lua.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4886860/3170.lua.tar.gz)

Issue was not reproduced on latest SDL develop - neither with fix nor without it.

There could be several reasons why issue is not reproduced in open source environment :
- Timings on Mobile\HMI communication may be different on CI/developer workstation and real hardware
- QNX platform specific issues may trigger this case
- Different CPU resources available on HU, and developer/CI.

Anyway script can be used to check use case on real hardware with Remote ATF feature.
And we think proposed fix should be included since it solves 'potential' problem.

theresalech, 2020-07-08:
@dboltovskyi, thank you for the additional information.  We'll await Luxoft's confirmation of PR #3375 being ready for Livio review.

PR #3375 (2020-09-02): Fix dangerous work with app commands

Fixes #3170

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Tested on SYNC version of SDL
For open source version will be created test script. 

### Summary
There was related to a thread safety violation
while working with application internal commands
array. This array is a thread safe itself as it is
protected with mutexes, however FindCommand()
function is returing a raw pointer to an internal
array element so the external component is able to
get access to this element at any time even without
locking the [... cut]