smartdevicelink/sdl_core #3171: CCB: SmartDeviceLink crash in application_manager with SIGSEGV

LitvinenkoIra, 2019-12-06:
### Bug Report
SmartDeviceLink crash in application_manager with SIGSEGV

### Detailed analysis:
There is a potential race condition in the remove_observer method, the observers_event_ map was not protected when calling EventDispatcherImpl::remove_observer(EventObserver& observer).

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile
2. Register app
3. Perform multiple RPC scenarios
4. Observe behavior

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashed
[bt.txt](https://github.com/smartdevicelink/sdl_core/files/3932187/bt.txt)

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a top priority for Toyota. Does this occur with the reproduction steps described? Also, does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@LitvinenkoIra, can you please advise if the Luxoft team is seeing this issue in Core 6.1?

LuxoftAKutsan, 2020-05-19:
Issue reproduced on SYNC version of SDL.
On open source version of SDL issue may be non reproducible. 
Script will be created. 
Fix for SYNC in incompatible with 6.1 
Probably issue may be fixed indirectly with aa8ad39cc936c7de63818d4307d8ac6a4dff5530

theresalech, 2020-05-19:
@LuxoftAKutsan, thank you!  We will wait for confirmation if the issue is reproducible in the open source based on the findings of your script.

dboltovskyi, 2020-07-07:
@theresalech Please notice,

Information provided in description is not enough to create exact sequence of actions in script. 
Some of the sequences are implelented in: [3171.lua.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4886957/3171.lua.tar.gz)

However we were not able to reproduce described SDL crash on latest develop.

theresalech, 2020-07-07:
@dboltovskyi, thank you for the additional information.  We will close this issue as not reproducible.

closed 2020-07-07 by theresalech