smartdevicelink/sdl_core #3172: CCB: Fix SIGSEGV Crash subscribed_data

LitvinenkoIra, 2019-12-06:
### Bug Report
SmartDeviceLink crash with SIGSEGV

### Detailed analysis:
unprotected access to the VehicleInfoAppExtension::subscribed_data from different threads

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile
2. Register app
3. Perform multiple RPC scenarios and send OnVehicleDataNotification
4. Observe behavior

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashed
[bt.txt](https://github.com/smartdevicelink/sdl_core/files/3932241/bt.txt)

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a high priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

theresalech, 2020-05-21:
@LitvinenkoIra can you please advise if you are able to reproduce this issue in the latest Core release?

LuxoftAKutsan, 2020-05-27:
Fix form SYNC4 will be applied.
Test script will be created, but test script may pass on open  sdl_core even without fix. Issue related to race conditions and may be not reproducible on open sdl.

theresalech, 2020-05-27:
@LuxoftAKutsan, thank you for the additional information! Please let us know when a PR is ready for Livio's review.

PR #3526 (2020-10-06): Re-order rai logic for plugins
Fixes #3172 #3162 

This PR is **Ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Send on vehicle data notification at same time app registers. Core should not crash.

### Summary
Moves the call to add the newly registered app to the applications list later in the registration process. This ensures that an app has all of its necessary plugins added before it is accessible by the applications accessor.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/v [... cut]
closed 2020-10-06 by JackLivio