smartdevicelink/sdl_core #3173: CCB: SmartDeviceLink SIGSEGV crash in std::__1::__tree_

LitvinenkoIra, 2019-12-06:
### Bug Report
SmartDeviceLink SIGSEGV crash in std::__1::__tree_iterator

### Detailed analysis:
When one thread accessing data of capabilities other thread can delete the pointer. So it necessary to change it to shared pointers.

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile
2. Register app
3. Observe behavior

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashed
[bt.txt](https://github.com/smartdevicelink/sdl_core/files/3932264/bt.txt)

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a high priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

theresalech, 2020-05-21:
@LitvinenkoIra can you please advise if you are able to reproduce this issue in the latest Core release?

LuxoftAKutsan, 2020-05-27:
Fix from SYNC4 applicable for open sdl. 
Additional test will not be created. any test case contains steps t reproduce. 
Probably issue can't be reproduce in open sdl environment. 
Regression will be checked

theresalech, 2020-05-27:
@LuxoftAKutsan, thank you for the additional information! Please let us know when a PR is ready for Livio's review.

JackLivio, 2020-10-06:
Resolved by #3528

closed 2020-10-06 by JackLivio