smartdevicelink/sdl_core #3174: CCB: App not Un-Registering when trying to Force Close Nav App

LitvinenkoIra, 2019-12-06:
### Bug Report
App not Un-Registering when trying to Force Close Nav App 

### Detailed analysis:
1. Video_streaming_Lock acquired was not released when timeout and stop called from another thread happens at the same time
- When a new video packet arrives and when wakeupstreaming is called, "video_streaming_suspend" and stop_streaming locks are acquired and a new video suspend timer is created. When the new video suspend timer is created it will stop the existing timer its referencing and then starts a new one.
The timer stop call waits until the timer callback is exited. In this case, during the stop process of the timer, a timeout occured and then the timer callback was waiting on acquiring the lock "video_streaming_suspend" which was not yet released by wakeupstreaming call. This puts it in an deadlock scenario.
- Solution for this issue will be, scoping the lock to protect only the variables it was intended to protect and not the timer stop and start calls.
2. Resumption timeout callback was waiting on Policy_Manager_lock_ which was not released when acquired if resuming application is activated.
- When an resuming app was activated and at the same time the resumption callback is triggered, policy_manager_lock wont be acquired by the resumption callback as its waiting on sdlactivateapp function call to release the lock, this lock wont be released as sdlactivateapp will be waiting to acquire "queue_lock" in resumption context which is already acquired by the resumption timeout callback.
- Solution: sdlactivateapprequest will release the "policy_manager_lock_" before activating the application as it wont be needed and allows resumption callback to acquire the lock to process resumption state for other application
3. Crash observed when one thread accessing multi-frame data and other thread deleting the multi frame data due to disconnection
- When One thread is processing mutliframe, another thread can trigger unexpected disconnect and remove the multiframe, which causes the multi frame processing thread to crash. Providing synchronization will resolve this issue

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile
2. Register Nav app, start streaming
3. Force Close Nav App

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashed

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a top priority for Toyota. Does this occur with the reproduction steps described? Also, does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@LitvinenkoIra, can you please advise if the Luxoft team is seeing this issue in Core 6.1?

LuxoftAKutsan, 2020-05-19:
For now we do not have appropriate script to reproduce the issue.
It was reproduced in SYNC environment with manual testing and and reproduced not always. 
On open source sdl with linux environment and the issue may not be reproducible.
Any way script will be crated.  

The issue is partially fixed with db92580aa80c4caee7e8907274aec22660ea552d 

Fix for SYNC-SDL contains more code, that is not  available in 6.1. 
PR with ported fix : https://github.com/smartdevicelink/sdl_core/pull/3374

aderiabin, 2020-09-02:
@theresalech Please notice the following findings:

Script for provided use case has been created: [CCB-3174 test script](https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2449)
Issue was not reproduced on latest SDL develop - neither with fix nor without it.

There could be several reasons why issue is not reproduced in open source environment :
 - Timings on Mobile\HMI communication may be different on CI/developer workstation and real hardware
 - QNX platform specific issues may trigger this case
 - Different CPU resources available on HU, and developer/CI.

Anyway script can be used to check use case on real hardware with Remote ATF feature.
And we think proposed fix should be included since it solves 'potential' problem.

jacobkeeler, 2020-09-02:
Closing with the merge of #3374

PR #3374 (2020-09-02): App not Un-Registering when trying to Force Close Nav App
Fixes #3174

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Manual testing performed on SYNC version for SDL.
Script from https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2449 

### Summary

1. video_streaming_Lock acquired was not released when timeout and stop called from another thread happens at the same time
2. resumption timeout callback was waiting on Policy_Manager_lock_ which was not released, when acquired if resuming application is activated.
3. Crash observered when [... cut]
closed 2020-09-02 by jacobkeeler