smartdevicelink/sdl_core #3175: CCB: Fix mutex deadlock on VR.ChangeLanguage received

LitvinenkoIra, 2019-12-06:
### Bug Report
Mutex deadlock on VR.ChangeLanguage received

### Detailed analysis:
There is a problem that on receiving VR.ChangeLanguage SDL is holding application lock via accessor and tries to send some notifications from the same thread.
For sending notifications SDL should check policy permissions for that notification and hance, to acquire policy lock.
At the same time, PTU could happen in a separate thread and during PTU policy lock could be acquired. Also, during PTU SDL will try to acquire applications lock from this thread, however it is acquired by another thread which executing VR.ChangeLanguage request.
As a result, we receive thread deadlock. To avoid this deadlock, accessor in VR.ChangeLanguage should be scoped in the same manner as for UI and TTS parts.

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile
2. Trigger PTU
3. Send VR.ChangeLanguage

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashed

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a top priority for Toyota. Does this occur with the reproduction steps described? Also, does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@LitvinenkoIra, can you please advise if the Luxoft team is seeing this issue in Core 6.1?

alexkutsan, 2020-05-18:
Hi @theresalech this issue was fixed on SYNC without specific test cases. 
Code fixes potential problem. 
Appropriate code is not fixed in scope of 6.1. 
Fix now available on https://github.com/smartdevicelink/sdl_core/pull/3370 
Test cases will be created. 
But due to potential nature of the issue test cases may not fail on open sdl.

LuxoftAKutsan, 2020-09-02:
Full regression was checked on #3370. Results are similar to develop

jacobkeeler, 2020-09-03:
Closing with the merge of #3370

PR #3370 (2020-09-03): Fix mutex deadlock on VR.ChangeLanguage received
Fixes #3175

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan

Issue does not contain special test cases. Fix is potential.

### Summary
There was a problem that on receiving VR.ChangeLanguage SDL is holding application lock via accessor and tries to send some notifications from the same thread.
For sending notifications SDL should check policy permissions for that notification and hance, to acquire policy lock.
At the same time, PTU could happen in the separate thread and during PTU policy lock coul [... cut]
closed 2020-09-03 by jacobkeeler