smartdevicelink/sdl_core #3178: CCB: SDL crash on set_delegate on shutdown

LitvinenkoIra, 2019-12-06:
### Bug Report
SDL crash on set_delegate on shutdown 

### Detailed analysis:
The core crash shows evidence that something around set_delegate is
crashing, on the other hand ~MessageLoopThread is deleting the thread
and not assigning the thread pointer to NULL which is necessary to avoid
undefined behavior

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile
2. Perform Ign Off

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashed

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san  This issue is a top priority for Toyota. Does this occur with the reproduction steps described? Also, does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@LitvinenkoIra, can you please advise if the Luxoft team is seeing this issue in Core 6.1?

LuxoftAKutsan, 2020-05-20:
@theresalech
Issue reproduced on SYNC version of SDL.
On open source version of SDL issue may be non reproducible.
Script will be created.
Fix from SYNC adopted for 6.1 codebase  https://github.com/smartdevicelink/sdl_core/pull/3382

aderiabin, 2020-08-28:
@theresalech Please notice the following findings:

Script for provided use case has been created: [3178.lua.tar.gz](https://github.com/smartdevicelink/sdl_core/files/5141298/3178.lua.tar.gz)

Issue was not reproduced on latest SDL develop - neither with fix nor without it.

There could be several reasons why issue is not reproduced in open source environment :
 - Timings on Mobile\HMI communication may be different on CI/developer workstation and real hardware
 - QNX platform specific issues may trigger this case
 - Different CPU resources available on HU, and developer/CI.

Anyway script can be used to check use case on real hardware with Remote ATF feature.
And we think proposed fix should be included since it solves 'potential' problem.

LuxoftAKutsan, 2020-09-02:
The issue  is covered with a #3388 fix.

jacobkeeler, 2020-09-02:
Closing as covered by #3388

closed 2020-09-02 by jacobkeeler