smartdevicelink/sdl_core #3181: CCB: Use weak_ptr<WebsocketSession> instead of raw ptrs

LitvinenkoIra, 2019-12-06:
### Bug Report
Use weak_ptr<WebsocketSession> instead of raw ptrs 

### Detailed analysis:
It prevents possible SIGSEGVs.
It's bad practice to raw pointers to WebsocketSession in almost all cases.
For function arguments, pointers should be changed to references for preventing null pointer occurrence.
For containers pointers should be changed to weak_ptr for preventing
expired references.
Wrong inheritance for CMessageBrokerController shpuld be removed.

##### Reproduction Steps
1. TBD

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashed

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a high priority for Toyota. Could you please tell me reproduction steps? Also, Does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@LitvinenkoIra, can you please provide reproduction steps for this issue? If you could also advise if the Luxoft team is seeing this issue in Core 6.1, that would be most helpful!

LuxoftAKutsan, 2020-05-20:
This issue has ho reproduction steps.
It was found by the Hellgrind dynamic code analysis tool in the SYNC environment.
6.1 do not contain the fix and the issue is possible.
SYNC changes need to be ported on open sdl_core.

theresalech, 2020-05-20:
@LuxoftAKutsan thank you for the update. Please advise when a PR has been submitted.

ShobhitAd, 2020-10-02:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3490

PR #3490 (2020-10-02): Use weak_ptr<WebsocketSession> instead of raw ptrs
Fixes #3181 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by unit tests

### Summary
This commit makes no functional or insecure changes.
It only prevents possible SIGSEGVs.
Don't use raw pointers to WebsocketSession in almost all cases.
For function arguments, pointers changed to references for preventing null pointer occurrence.
For containers pointers changed to weak_ptr for preventing expired references.
Removed wrong inheritance for CMessageBroker

### Tasks Remaining:
- [x] [Test [... cut]
closed 2020-10-02 by ShobhitAd