smartdevicelink/sdl_core #3182: CCB: Use atomic bool in ApplicatinManager

LitvinenkoIra, 2019-12-06:
### Bug Report
Deadlock inside class "ApplicationManagerImpl"

### Detailed analysis:
Deadlock happens because of using two mutexes for one critical section
inside classes "ApplicationManagerImpl" and "ResumeCtrlImpl".
We are trying to lock one section by different mutexes.
Resolving, by copy set of applications into temporary set and use it
instead of guarded app set which returns by function 'applications()'.
Use atomic bool for flag 'is_stopping_' in class 'ApplicationManagerImpl', remove mutex 'stopping_application_mng_lock_' which needed for guarding of i's_stopping_'.

##### Reproduction Steps
1. TBD

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashed

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a high-priority for Toyota. Could you please tell me reproduction steps? Also, Does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@LitvinenkoIra, can you please provide reproduction steps for this issue? If you could also advise if the Luxoft team is seeing this issue in Core 6.1, that would be most helpful!

LuxoftAKutsan, 2020-05-20:
This issue has ho reproduction steps.
It was found by the Hellgrind dynamic code analysis tool in the SYNC environment.
6.1 do not contain the fix and issue is possible. 
SYNC changes need to be ported on open sdl_core.

theresalech, 2020-05-20:
@LuxoftAKutsan thank you for the update.  Please advise when a PR has been submitted.

AKalinich-Luxoft, 2020-09-04:
@theresalech fix has been submitted and ready for Livio review

iCollin, 2020-09-30:
Closed via #3491

PR #3491 (2020-09-30): Use atomic bool in ApplicationManager
Fixes #3182 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by unit tests

### Summary
Replace bool flag 'is_stopping_' in class ApplicationManagerImpl and remove mutex 'stopping_application_mng_lock_'.

### Tasks Remaining:
- [x] Test this fix

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-09-30 by iCollin