smartdevicelink/sdl_core #3184: CCB: wait for UsbHandler thread end in UsbHandler dtor

LitvinenkoIra, 2019-12-06:
### Bug Report
Data races related to UsbHandler

### Detailed analysis:
The Thread dedicated to handling libusb events should exit successfully on it's own. This guarantees that libusb_close() being called on all devices, as well as libusb_exit() call is properly sequenced.
Thread exits in recommended by libusb doc way:
  1. set exit flag
  2. deregister hotplug callbacks, which will wake up
     libusb_handle_events() once again.
The way the Thread being joined should be changed. It should ensure that join() is called with kNoStop flag, which prevents force stop using pthread_cancel().

##### Reproduction Steps
1. TBD

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashed

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

AKalinich-Luxoft, 2020-09-04:
@theresalech fix has been submitted and ready for Livio review

ShobhitAd, 2020-10-06:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3492

PR #3492 (2020-10-06): Fix/usb handler thread safe stop
Fixes #3184 and #3188 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by unit tests

### Summary
- Wait for UsbHandler thread end in UsbHandler dtor 
- Using atomic bool for `shutdown_requested_` flag

### Tasks Remaining:
- [ ] Test this fix

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-10-06 by ShobhitAd