smartdevicelink/sdl_core #3185: CCB: Make refactoring in mb_controller.cc

LitvinenkoIra, 2019-12-06:
### Bug Report
Race condition in mb_controller.cc 

### Detailed analysis:
Possible race condition in case of using assignment operator
with atomic variable 'shutdown_', fix via using 'atomic_exchange'.
Incorrect sequence of closing boost asio objects.
Add member function into CMessageBrokerController for correct close sequence.

##### Reproduction Steps
1. TBD

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashed

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a high priority for Toyota. Could you please tell me reproduction steps? Also, Does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@LitvinenkoIra, can you please provide reproduction steps for this issue? If you could also advise if the Luxoft team is seeing this issue in Core 6.1, that would be most helpful!

LuxoftAKutsan, 2020-05-20:
This issue has ho reproduction steps.
 It was found by the Hellgrind dynamic code analysis tool in the SYNC environment.
 PR to open now available https://github.com/smartdevicelink/sdl_core/pull/3381
 Regression with automated tests will be performed.

ShobhitAd, 2020-07-13:
Closed via #3381

PR #3381 (2020-07-13): Make refactoring in mb_controller.cc
Fixes #3185

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Automated regression will be performed
### Summary

Fix race condition regarding Helgrind issue.
Error 9 : Possible race condition in case of using assignment operator
with atomic variable 'shutdown_', fix via using 'atomic_exchange'.
Error 10: Incorrect sequence of closing boost asio objects.
Add member function into CMessageBrokerController for correct close sequence.

### Tasks Remaining:
- [ ] Luxoft internal review
- [ ] Automated regression

### [... cut]
closed 2020-07-13 by ShobhitAd