smartdevicelink/sdl_core #3186: CCB: Fix trying to unlock mutex not held by any thread

LitvinenkoIra, 2019-12-06:
### Bug Report
Data races related to MessageQueue

### Detailed analysis:
Mutex released before call Broadcast.

##### Reproduction Steps
1. TBD

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashed

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san  This issue is a high priority for Toyota. Could you please tell me reproduction steps?  Also, Does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@LitvinenkoIra, can you please provide reproduction steps for this issue? If you could also advise if the Luxoft team is seeing this issue in Core 6.1, that would be most helpful!

LuxoftAKutsan, 2020-05-20:
This issue has ho reproduction steps.
 It was found by the Hellgrind dynamic code analysis tool in the SYNC environment.
 PR to open now available https://github.com/smartdevicelink/sdl_core/pull/3380
 Regression with automated tests will be performed.

PR #3380 (2020-09-03): Fix trying to unlock mutex not held by any thread

Fixes #3186

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Regression automated testing

### Summary

Fix trying to unlock mutex not held by any thread
Helgrind considers that this issue is an error because we release mutex before call Broadcast.

- [ ] Luxoft internal review 
- [ ] Regression automated testing

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-09-03 by JackLivio