smartdevicelink/sdl_core #3189: CCB: Fix dead lock in usb connection

LitvinenkoIra, 2019-12-06:
### Bug Report
Data races related to UsbConnection

### Detailed analysis:
In a while called method locks mutex for protecting data and if USB connection is failed it means that method will call and will try to lock the same mutex again. That's why this mutex should be recursive.

##### Reproduction Steps
1. TBD

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashed

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

AKalinich-Luxoft, 2020-09-04:
@theresalech fix has been submitted and ready for Livio review

ShobhitAd, 2020-10-05:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3494

PR #3494 (2020-10-05): Fix deadlock in usb connection
Fixes #3189 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by unit tests

### Summary
In a while called method  locks mutex for protecting data and if USB connection is failed it means that method will call and will try to lock the same mutex again. That's why this mutex should be recursive.

### Tasks Remaining:
- [ ] Test this fix

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-10-05 by ShobhitAd