smartdevicelink/sdl_core #3190: CCB: Fix deadlock in timer.cc

LitvinenkoIra, 2019-12-06:
### Bug Report
Deadlock in timer.cc 

### Detailed analysis:
In Timer::TimerDelegate::exitThreadMain() lock calls two times for Lock, second lock should be was removed.

##### Reproduction Steps
1. TBD

##### Expected Behavior
SDL should not crash

##### Observed Behavior
SDL crashed

#### OS & Version Information
* OS/Version : SYNC4
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/5.0.1 
* Testing Against: SYNC4 Applink

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san  This issue is a high-priority for Toyota. So I would like to know the status of this issue,  could you please tell me reproduction steps? Also, does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release.  This planning will take place in June.
@LitvinenkoIra, can you please provide reproduction steps for this issue?  If you could also advise if the Luxoft team is seeing this issue in Core 6.1, that would be most helpful!

LuxoftAKutsan, 2020-05-20:
This issue have ho reproduction steps. 
It was found by Hellgrind dynamic code analysis tool in SYNC environment. 
PR to open now available https://github.com/smartdevicelink/sdl_core/pull/3379 
Regression with automated tests will be performed.

jacobkeeler, 2020-06-16:
Closing with the merge of #3379

PR #3379 (2020-06-16): Fix deadlock in timer.cc

Fixes #3190

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Regression testing should be performed

### Summary

Fix deadlock in timer.cc call lock two times for Lock,
second lock was removed, cause using of it hadn't any benefits.
Regarding of Helgrind report.

### Tasks Remaining:
- [ ] Luxoft Intenal review 
- [ ] Regression testing

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-06-16 by jacobkeeler