smartdevicelink/sdl_core #3193: [SDL 0261] New vehicle data WindowStatus

theresalech, 2019-12-12:
Proposal: [New vehicle data WindowStatus](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0261-New-vehicle-data-WindowStatus.md)

> This proposal is to add new vehicle data `WindowStatus` to know the position of windows in a vehicle.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/869

Steering Committee Decision:

>The Steering Committee voted to accept this proposal with the revision to `WindowStatus` described in [this comment](https://github.com/smartdevicelink/sdl_evolution/issues/869#issuecomment-564050172).

> The proposal .md file was updated to reflect these revisions on 12/12/19.

theresalech, 2020-06-11:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0261 Revisions - New vehicle data WindowStatus](https://github.com/smartdevicelink/sdl_evolution/issues/1028).  Revisions adding `mandatory=true` attribute to `approximatePosition` and `deviation`, and changing `minValue/MaxValue` attributes to `minvalue/maxvalue`.  Please reference the issue and proposal markdown file for revisions details.

ShobhitAd, 2020-08-07:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3391

PR #3391 (2020-08-07): SDL Core implementation of feature - 0261 New vehicle data WindowStatus
Implements/Fixes #3193 

This PR is **ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan
Unit testing + ATF testing 

### Summary
1. Added new `WindowStatus` and `WindowState` data types to API.
2. Updated VehicleData RPCs with new parameters (Subscribe/Unsubscribe/Get/OnVehicleData).
3. Updated unit tests.
4. Updated `sdl_preloaded_pt.json` file.

### CLA
- [X] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-08-07 by ShobhitAd