smartdevicelink/sdl_core #3197: [SDL 0240] WebEngine support for SDL JavaScript

theresalech, 2019-12-19:
Proposal: [WebEngine support for SDL JavaScript](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0240-sdl-js-pwa.md)

> This proposal is adding a new transport to the SDL JavaScript library to support (progressive) web apps to run on a WebEngine or a browser.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/767

Steering Committee Decision:

>The Steering Committee voted to accept this proposal with the following revisions:
- When formatting `locales` in the `manifest.json` file, use standard [(RFC 5646)](https://tools.ietf.org/html/rfc5646), and convert locale strings when necessary.
- Do not specify the location of the manifest's <script> tag placement within the proposal.
- All revisions included in this PR #888 
> The proposal .md file was updated to reflect these revisions on 12/19/19.

dboltovskyi, 2020-01-27:
@theresalech 
Along with the current issue for `sdl_core` we have prepared an updates for `sdl_atf` and `sdl_atf_test_scripts` in order to validate the feature with automated tests.
So we would like to ask you to create 2 additional issues in these repositories as well.

dboltovskyi, 2020-01-27:
@theresalech Please notice one more issue for `sdl_hmi_integration_guidelines` is also required.

theresalech, 2020-01-27:
@dboltovskyi these have been created, thanks!
- https://github.com/smartdevicelink/sdl_hmi_integration_guidelines/issues/188
- https://github.com/smartdevicelink/sdl_atf_test_scripts/issues/2322
- https://github.com/smartdevicelink/sdl_atf/issues/182

theresalech, 2020-02-12:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0240 Revisions - WebEngine support for SDL JavaScript](smartdevicelink/sdl_evolution#923). These revisions change the `appID` parameter within `GetAppProperties` to `policyAppID` and add a minlength of `1` to the param. Please see the proposal markdown file for more details.

closed 2020-02-27 by JackLivio