smartdevicelink/sdl_core #3204: SDL sends OnAppInterfaceUnregistered with the reason IGNITION_OFF during FACTORY_DEFAULTS exit

GetmanetsIrina, 2019-12-28:
### Bug Report
SDL sends OnAppInterfaceUnregistered with the reason `IGNITION_OFF` during `FACTORY_DEFAULTS` exit

##### Reproduction Steps
1. SDL and HMI are started
2. App is registered
3. HMI sends `BC.OnExitAllApplications(FACTORY_DEFAULTS)`

##### Expected Behavior
SDL unregisters app and sends OnAppInterfaceUnregistered(FACTORY_DEFAULTS) to it.

##### Observed Behavior
SDL sends OnAppInterfaceUnregistered(IGNITION_OFF) to mobile application

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version:https://github.com/smartdevicelink/sdl_core/commit/e137556512215ded94228e33d21ab4c3e1af3017
* ATF: develop
* Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/4_5/1225_FACTORY_DEFAULTS.lua

##### Logs
[SDL_and_ATF_logs.zip](https://github.com/smartdevicelink/sdl_core/files/4007004/SDL_and_ATF_logs.zip)

LitvinenkoIra, 2020-03-03:
https://github.com/smartdevicelink/sdl_atf_test_scripts/issues/2355

E-SAITO-TMC, 2020-04-20:
@theresalech -san This issue is a medium priority for Toyota. I would like know if this issue occur in the latest version. And could you please include this issue in your plan? Thank you.

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is medium priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release.

@GetmanetsIrina, can you please advise if you are seeing this issue when using Core 6.1?

GetmanetsIrina, 2020-04-23:
@theresalech, this issue is still actual for Core 6.1(https://github.com/smartdevicelink/sdl_core/commit/a85ef58358c456f69fee93addb7066dc02ecae48)

iCollin, 2020-05-18:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3274

PR #3274 (2020-05-18): Change reason to IGNITION_OFF only if reason wasn't defined before
Fixes #3204 

This PR is **[ready]** for review.

### Risk
This PR makes **[no]** API changes.

### Testing Plan
Covered within existing unit tests and ATF tests.
During regression incorrect script was found, this fact was reflected in corresponding issue https://github.com/smartdevicelink/sdl_atf_test_scripts/issues/2355, and this problem is solved in pull request https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2354.

### Summary
This pull request is intended to fix issue with changing reason from FACTORY_DEFAULTS [... cut]
closed 2020-05-18 by iCollin