smartdevicelink/sdl_core #3207: SDL is unable to update Pixel density and Scale through OnSystemCapabilityUpdated

dboltovskyi, 2020-01-07:
### Bug Report
SDL is unable to update Pixel density and Scale parameters through `OnSystemCapabilityUpdated` notification

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile
=> During init process HMI provides to SDL default values for Pixel density and Scale parameters in `UI.GetCapabilities`
2. Register App
3. Send `BC.OnSystemCapabilityUpdated` from HMI with new values for parameters
4. Send `GetSystemCapability(VIDEO_STREAMING)` request from App 

##### Expected Behavior
SDL updates the values of Pixel density and Scale parameters and transfer them in `OnSystemCapabilityUpdated` notification to App.
SDL provides updated values in `GetSystemCapability` response to App.

##### Observed Behavior
SDL doesn't update the values and provides default ones

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (0fff71b)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - Script: 
 - Report & logs:

dboltovskyi, 2020-01-08:
@theresalech Please triage this issue

theresalech, 2020-01-08:
@dboltovskyi this issue has not yet been raised to the Steering Committee as priority for the upcoming Core release. We can work to include if time allows, but would recommend raising as priority in the Steering Committee meeting if it's imperative the issue be included. Thanks!

JackLivio, 2020-01-13:
Are you sure this is a bug? The OnBCSystemCapabilityUpdatedNotificationFromHMI class only looks for capability types DISPLAYS and REMOTE_CONTROL. I am leaning more towards that this is a feature request instead of a bug.

dboltovskyi, 2020-01-14:
@theresalech @JackLivio Please notice a new proposal for SDL will be created regarding possibility to update `VIDEO_STREAMING` capabilities through `OnSystemCapabilityUpdated` notification.

closed 2020-01-14 by dboltovskyi