smartdevicelink/sdl_core #3232: SDLCore failed to retry updating the Policy Table

zhouxin627, 2020-02-06:
### Bug Report
SDLCore failed to retry updating the Policy Table

##### Preconditions
Connect the head unit to any Android/iOS phone

##### Reproduction Steps
1. The first time of updating Policy Table is failed that caused by network problems
2. Fix the network problems soon.

##### Expected Behavior
SDLCore retries updating Policy Table is successful.

##### Observed Behavior
SDLCore retries updating Policy Table is failed.
It sends the correct URL to the phone for the first time of updating the Policy Table.
But when a retry is requested, it sends an empty URL to the phone,
so it certainly failed to update the Policy Table.

##### OS & Version Information
* SDL Core Version: SDL Core 6.0.1
* Testing Against:SDL Core 6.0.1 + HMI running on our development environment.

##### Occurrence:
100%

##### Recovery Method:
None

jacobkeeler, 2020-03-17:
@zhouxin627 testing with the SDL HMI, this seems to work fine. The HMI sends a url in it's `OnSystemRequest` during the retry sequence, which gets passed down to the application. What policy mode are you using?

I'll note that I am testing with the `release/6.1.0` (and `develop` for the HMI) rather than release 6.0.1, so it's possible that it might have been fixed recently. Could you test against these branches as well?

zhouxin627, 2020-03-19:
@jacobkeeler Hi, thanks for your advice. We are using the `regular_PROPRIETARY` mode and I would test with 6.1.0. Could you please give me more information about which PR or commit fixed this issue.

jacobkeeler, 2020-03-19:
@zhouxin627 I do not know what commit this issue might have been fixed in, I just can't reproduce the issue in that branch. If you test your setup against the `release/6.1.0` branch, we can determine for certain whether this is an actual Core issue or an HMI issue. You are using your own custom HMI, correct?

zhouxin627, 2020-03-20:
@jacobkeeler Yes, I got it! We are using our own custom HMI.

theresalech, 2020-03-23:
@zhouxin627 thanks for letting us know.  Please advise on the results of your testing against the `release/6.1.0` branch at your earliest convenience.  Thanks!

jacobkeeler, 2020-03-30:
@zhouxin627 My apologies, it seems that a difference in my environment was the reason why I couldn't reproduce the issue, I am able to now. Reviewing your fix now

jacobkeeler, 2020-04-20:
Closing with the merge of #3329

PR #3329 (2020-04-20): Fix PROPRIETARY retry sequence
Fixes #3232

This PR is **ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan
1. Edit the policy table module config:
```
          ...
          "timeout_after_x_seconds": 5,
            "seconds_between_retries": [
                1,
                1,
                1,
                1,
                1
            ],
            "endpoints": {
                "0x07": {
                    "default": [
                        "http://smartdevicelink.com/api/1/policies/example1"
                    ],
                    "867 [... cut]
closed 2020-04-20 by jacobkeeler