smartdevicelink/sdl_core #3233: SDL failed to start video streaming due to App switching on iPhone(connection via BT+WIFI)

zhouxin627, 2020-02-06:
### Bug Report
SDL failed to start video streaming due to App switching on iPhone(connection via BT+WIFI)

### Preconditions
1. Connect the head unit to iPhone via BT+WiFi
2. Register 2 iOS apps:
・App1: navigation App
・App2: navigation App

##### Reproduction Steps
1. Activate App1 on HU
2. App1 starts video streaming
3. Switch App1 to the background on iPhone
4. Launch App2 on iPhone
5. Activate App2 on HU
6. App2 failed to start video streaming

##### Expected Behavior
App2 starts video streaming successfully.

##### Observed Behavior
A timing issue due to that VPM terminate operation and playing video is being executed by separate threads (these threads run simultaneously).
1. When switching App1 to the background on iPhone, WIFISession will be closed according to the processing logic of iOS, Proxy will send VPM termination request to SDLCore.
2. SDLCore will call `OnServiceEndedCallback(11)` to terminate VPM.
3. At this time, SDLCore received the video data that still not be sent completed.
Then SDLCore will think there's still has video data to play, and restarts VPM that shouldn't be done.
4. Eventually, App1 is switched to the background on iPhone but SDLCore still sends `StartStream` to HU that shouldn't be done, therefore, the status of VPM becomes chaotic.
5. App2 failed to start video streaming.

##### OS & Version Information
* iOS Version:10.2
* Proxy Version: 6.4.1
* Testing Against: SDL Core 6.0.1 + HMI running on our development environment.

##### Occurrence:
80%

##### Recovery Method:
ACC OFF/ON

mrapitis, 2020-03-23:
@theresalech would it be possible to prioritize review of this bug and plan for an upcoming sdl core release?

jacobkeeler, 2020-03-23:
@zhouxin627 point of clarification, could you define VPM?

zhouxin627, 2020-03-25:
@jacobkeeler Hi, “VPM” is the abbreviation of “Video streaming Projection Mode”. Actually, “VPM“  has the same meaning as video streaming.

JackLivio, 2020-04-03:
Fixed with merge #3290

PR #3290 (2020-04-03): Fix for video stream failed by switched of connection via BT+WIFI
Fixes #3233 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by Unit tests.

### Summary
**Reason**
1. When switching App1 to the background on iPhone, WIFI Session will be closed according to the processing logic of iOS, Trigger inner VPM termination request to SDLCore.
2. SDLCore will call `OnServiceEndedCallback(11)` to terminate VPM.
3. At this time, SDLCore received the video data that still not be sent completed. Then SDLCore will think there's still has video data to [... cut]
closed 2020-04-03 by JackLivio