smartdevicelink/sdl_core #3237: Redundant SDL.OnStatusUpdate(UPDATE_NEEDED) notification is sent once device is consented

dboltovskyi, 2020-02-06:
### Bug Report
Redundant SDL.OnStatusUpdate(UPDATE_NEEDED) notification is sent once device is consented

##### Occurrences
95%

##### Reproduction Steps
1. Make sure SDL is built in `EXTERNAL_PROPRIETARY` policy mode
2. Start SDL, HMI, connect Mobile device
3. Register new application
=> SDL sends SDL.OnStatusUpdate(UPDATE_NEEDED) notification to HMI
4. Activate application and consent device

##### Expected Behavior
SDL sends SDL.OnStatusUpdate(UPDATING) notification to HMI

##### Observed Behavior
SDL sends redundant SDL.OnStatusUpdate(UPDATE_NEEDED) notification to HMI

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (4f17e76233)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
  - Script: [308_ATF_Check_app_registration_language_gui](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/Validation_of_PolicyTables/308_ATF_Check_app_registration_language_gui.lua)
  - Report and logs: [2020-02-06_16-36-05.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4167882/2020-02-06_16-36-05.tar.gz)

dboltovskyi, 2020-02-06:
Please notice this is a regression issue which was introduced by [0248-hmi-ptu-support](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0248-hmi-ptu-support.md) feature.
It is not reproduced at e9709e6701

dboltovskyi, 2020-02-12:
Issue has the same root cause as #3136 hence could be fixed by #3208

dboltovskyi, 2020-02-12:
@theresalech Please triage this issue

JackLivio, 2020-02-28:
Fixed by #3208

closed 2020-02-28 by JackLivio