smartdevicelink/sdl_core #3239: SDL sends BC.PolicyUpdate twice

dboltovskyi, 2020-02-07:
### Bug Report
SDL sends BC.PolicyUpdate twice

##### Reproduction Steps
1. Make sure SDL is built in `PROPRIETARY` policy mode
2. Start SDL, HMI, connect Mobile device, register new Application
=> PTU is started
3. Finish PTU successfully
=> SDL sends `VI.GetVehicleData(odometer = true)` request to HMI
4. Respond from HMI with some value, e.g. `1000`
5. Send `VI.OnVehicleData` with other `odometer` value which is greater than `exchange_after_x_kilometers`, e.g. `3000`
=> PTU is started
=> SDL sends `BC.PolicyUpdate` request to HMI
6. Send successful response from HMI

##### Expected Behavior
SDL doesn't send 2nd `BC.PolicyUpdate` and continue with PTU sequence

##### Observed Behavior
SDL sends 2nd `BC.PolicyUpdate` request to HMI

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (4f17e76)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
  - Script: [110_ATF_PTU_Trigger_kilometers_PROPRIETARY](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/web_engine_support/test_scripts/Policies/build_options/110_ATF_PTU_Trigger_kilometers_PROPRIETARY.lua)
  - Reports and logs:

dboltovskyi, 2020-02-07:
Please notice this is a regression issue which was introduced by [0248-hmi-ptu-support](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0248-hmi-ptu-support.md) feature.
It is not reproduced at 49f5b211ca

dboltovskyi, 2020-02-12:
@theresalech Please triage this issue

theresalech, 2020-02-12:
@dboltovskyi please let us know when the PR to resolve this regression issue is ready for Livio review, thanks!

dboltovskyi, 2020-02-18:
@theresalech Please notice PRs in `core` and `scripts` are ready for PM review.

PR #3242 (2020-02-20): Adapt fix #3203 for PROPRIETARY flow
Fixes #3239 

This PR is **[ready]** for review.

### Risk
This PR makes **[no]** API changes.

### Testing Plan
ATF scripts

### Summary
The problem described in #3138, became apparent for PROPRIETARY flow as well. This PR applies fix from mentioned issue for PROPRIETARY flow.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-02-20 by JackLivio