smartdevicelink/sdl_core #3246: SDL does not support attribute "until" for any element in Mobile_API

ydementieiev, 2020-02-18:
### Bug Report
During implementation [Proposal: Refactor Fuel Information Related Vehicle Data](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0256-Refactor-Fuel-Information-Related-Vehicle-Data.md)  was found that individual elements do not support attribute `until` in `Mobile_API`.

### Reproduction Steps
Update enum `VehicleDataType` adding `until` for individual elements:
```
<element name="VEHICLEDATA_FUELLEVEL" until="6.0"/>
<element name="VEHICLEDATA_FUELLEVEL_STATE" until="6.0"/>
```

### Expected Behavior
According to [Mobile API versioning](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0089-mobile-api-versioning.md) The attribute `since` and `until` should be used for any element (enum, element, function, struct, param) in the `Mobile_API`.

### Observed Behavior
Attribute "until" does not support for any element in Mobile_API

### OS & Version Information
OS/Version: [Ubuntu 18.0]
SDL Core Version: [Core version](https://github.com/smartdevicelink/sdl_core/commit/b281c67f167f385aabbb232333ddc8cef65172d7)

ShobhitAd, 2020-02-25:
@ydementieiev The until param change should also be added to the https://github.com/smartdevicelink/rpc_spec/blob/develop/MOBILE_API.xsd file

ShobhitAd, 2020-02-26:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3249

ShobhitAd, 2020-03-02:
Found additional attributes(`deprecated` and `removed`) which were also missing in the schema. Creating a new fix PR

ShobhitAd, 2020-03-02:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3266

PR #3249 (2020-02-26): Added missing until param for enumeration elements
Fixes #3246

This PR is **[ready]** for review.

### Risk
This PR makes **[minor]** API changes.

### Summary
Added missing attribute "until" for support it in any enum element in Mobile_API.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
PR #3266 (2020-03-02): Add missing attributes for API elements in the schema
Fixes #3246 

This PR is **ready** for review.

### Risk
This PR makes **minor** API changes.

### Summary
Added the the missing `deprecated` attribute for elementType and `removed` attribute for all API elements

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-03-02 by ShobhitAd