smartdevicelink/sdl_core #3267: Sometimes SDL doesn't send OnSystemRequest(LOCK_SCREEN_ICON_URL)

dboltovskyi, 2020-03-02:
### Bug Report
Sometimes SDL doesn't send `OnSystemRequest(LOCK_SCREEN_ICON_URL)`

##### Occurrence
Sometimes (0.5%)

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile device, register new Application
3. Check SDL sends `OnSystemRequest(LOCK_SCREEN_ICON_URL)`

##### Expected Behavior
Notification is sent

##### Observed Behavior
Notification is not sent due to missing `url` reason:
```
ERROR [29 Feb 2020 01:37:22,598][0x7f4faa361700][Commands] /home/developer/sdl/sdl_core/src/components/application_manager/rpc_plugins/sdl_rpc_plugin/src/commands/mobile/on_system_request_notification.cc:155 commands::mobile::OnSystemRequestNotification::Run: discarding LOCK_SCREEN_ICON_URL request without URL
```

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (ce12181)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App

dboltovskyi, 2020-03-03:
Issue became reproducible after merge fix for https://github.com/smartdevicelink/sdl_core/issues/2932

dboltovskyi, 2020-03-05:
@theresalech Please triage this issue.
Updates for `core` and `scripts` are ready for PM review.

iCollin, 2020-03-12:
Do you really see this issue occur 0.5% of runs? I have now run [3267_1_RAI_OnSystemRequest_LOCK_SCREEN_ICON_URL.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/6eb8406f4762eab4337dc76e3890c9de7e700837/test_scripts/Defects/6_1/3267_1_RAI_OnSystemRequest_LOCK_SCREEN_ICON_URL.lua) 50 times (each run tests 200 times) and I have yet to see a failure.

Can you give me any hint on how to reproduce this more easily? For example will this occur more often is I limit the processing power/memory/IO bandwidth core has available?

iCollin, 2020-03-13:
Closed via #3271

PR #3271 (2020-03-13): Fix OnSystemRequest(LOCK_SCREEN_ICON_URL) sending
Fixes #3267 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered with ATF test script

### Summary
There was noticed an issue that sometimes Policy Manager component is returning an empty url for lock screen icon. The root cause of that is the data races between thread which is sending `OnSystemRequest` notification and backupper thread which overwrites data in the database. As these two threads does not synchronized, there could be a possible situation that backuper thread erased data from `e [... cut]
closed 2020-03-13 by iCollin