smartdevicelink/sdl_core #3275: Sometimes SDL doesn't consider timeouts during streaming switching

dboltovskyi, 2020-03-04:
### Bug Report
Sometimes SDL doesn't consider timeouts during streaming switching

##### Occurrence
Sometimes (0.5%)

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile device, register 2 new streaming applications, e.g. 
 - App_1: `PROJECTION`
 - App_2: `NAVIGATION`
2. Deactivate App_1
=> streaming still continue since app has `STREAMABLE` state
3. Activate App_2

##### Expected Behavior
SDL does:
 - send OnHMIStatus(FULL, STREAMABLE) notification to App_2
 - send OnHMIStatus(LIMITED, NOT_STREAMABLE) notification to App_1
 - send Navi.OnVideoDataStreaming(false) notification to HMI
 - start 1st 'StopStreamingTimeout' timeout

Once timeout is expired SDL does:
 - send Navi.StopStream(App_1) request to HMI
 - send EndService(VIDEO) request to App_1
 - start 2nd 'StopStreamingTimeout' timeout

##### Observed Behavior
SDL does not consider 'StopStreamingTimeout' and send all the messages at once

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (912e3cf786)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - Script: [3139_02_Stop_streaming_gracefully_2nd_timer](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/6_1/3139_02_Stop_streaming_gracefully_2nd_timer.lua)
 - Report and Logs: [2020-03-04_13-10-27.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4288989/2020-03-04_13-10-27.tar.gz)

AKalinich-Luxoft, 2020-03-05:
For understanding of the root cause of that issue, some things should be described here.

Technically, SDL do the following actions after activation of `App_2` at step `App_2_activated__App_1_is_not_unregistered`:
1. Changes HMI state (e.g. `STREAMABLE` -> `NOT_STREAMABLE`) internally
```
DEBUG [04 Mar 2020 18:14:43,754][0x7f3777fff700][StateControllerImpl] /home/developer/sdl/sdl_core/src/components/application_manager/src/state_controller_impl.cc:862 StateControllerImpl::OnStateChanged: Window #0 old state: HMIState(app id:65537, hmi app id:652436598, window_type:0, state:REGULAR, hmi_level:1, audio:0, video:0, context:0)

DEBUG [04 Mar 2020 18:14:43,755][0x7f3777fff700][StateControllerImpl] /home/developer/sdl/sdl_core/src/components/application_manager/src/state_controller_impl.cc:864 StateControllerImpl::OnStateChanged: Window #0 new state: HMIState(app id:65537, hmi app id:652436598, window_type:0, state:VIDEO_STREAMING, hmi_level:1, audio:0, video:1, context:0)
```

2. Sends `OnHMIStatus` notification to affected application
```
DEBUG [04 Mar 2020 18:14:43,761][0x7f37777fe700][ApplicationManager] /home/developer/sdl/sdl_core/src/components/application_manager/src/mobile_message_handler.cc:122 RawMessage* application_manager::MobileMessageHandler::HandleOutgoingMessageProtocol: Outgoing RPC_INFO: 1, Notification, 32768, -1, {
	"audioStreamingState" : "AUDIBLE",
	"hmiLevel" : "LIMITED",
	"systemContext" : "MAIN",
	"videoStreamingState" : "NOT_STREAMABLE",
	"windowID" : 0
}
```

As can be seen, SDL has some time period for about **10ms** between actions 1 and 2. Let's call it **"state transition period"**.

The issue is happening, when SDL receives audio/video streaming data during the "state transition period":
```
TRACE [04 Mar 2020 18:14:43,757][0x7f37a68af700][MediaManager] /home/developer/sdl/sdl_core/src/components/media_manager/src/media_manager_impl.cc:304 MediaManagerImpl::OnMessageReceived: Enter
TRACE [04 Mar 2020 18:14:43,757][0x7f37a68af700][ApplicationManager] /home/developer/sdl/sdl_core/src/components/application_manager/src/application_manager_impl.cc:3448 ApplicationManagerImpl::CanAppStream: Enter
TRACE [04 Mar 2020 18:14:43,757][0x7f37a68af700][ApplicationManager] /home/developer/sdl/sdl_core/src/components/application_manager/src/application_manager_impl.cc:3472 ApplicationManagerImpl::ForbidStreaming: Enter

DEBUG [04 Mar 2020 18:14:43,757][0x7f37a68af700][ApplicationManager] /home/developer/sdl/sdl_core/src/components/application_manager/src/application_manager_impl.cc:3563 ApplicationManagerImpl::EndNaviServices: Going to end video service
```

The problem is that `CanAppStream` returns `false` right after action 1, because technically `App_1` has already a new state, however application itself was not notified yet (in action 2).
As a result, SDL considers that `App_1` is streaming when it shouldn't:
```
ERROR [04 Mar 2020 18:14:43,773][0x7f37a68af700][MediaManager] /home/developer/sdl/sdl_core/src/components/media_manager/src/media_manager_impl.cc:318 MediaManagerImpl::OnMessageReceived: The application trying to stream when it should not.
```

This is a reason why SDL sends EndService and notifications right after receiving streaming message without any timeout consideration.

Looks like some synchronization between "state transition period" and `CanAppStream` function required.

dboltovskyi, 2020-03-18:
@theresalech Please triage this issue. This one is related to already fixed streaming issues in scope of 6.1. 
Please notice linked PR in `sdl_core` is ready for PM review.

jacobkeeler, 2020-03-19:
Closing with the merge of #3293

PR #3293 (2020-03-19): Fix/consider timeouts during streaming switching
Fixes #3275 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF Script: [3139_02_Stop_streaming_gracefully_2nd_timer](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/6_1/3139_02_Stop_streaming_gracefully_2nd_timer.lua)

### Summary
The issue is happening, when SDL receives audio/video streaming data during changes HMI state (e.g. STREAMABLE -> NOT_STREAMABLE) internally. Earlier SDL considered only HMI_Level, but now after fix: [Fix SDL app stream processi [... cut]
closed 2020-03-19 by jacobkeeler