smartdevicelink/sdl_core #3279: SDL does not send BC.PolicyUpdate to HMI if no apps are registered

ShobhitAd, 2020-03-09:
### Bug Report
Currently, if there are no registered apps, triggering a PTU results in the HMI sending a `BasicCommunication.OnSystemRequest` with `fileName: null`. The PTS filename is never set in the HMI because SDL does not send a `BasicCommunication.PolicyUpdate` request(which contains the PTS path) to the HMI. This prevents us from being able to use the HMI vehicle modem to handle the PTU if no apps are registered.

##### Reproduction Steps
Using the sdl hmi with external proprietary checked:
1. Build core with external proprietary
2. Start the policy server
3. Modify the `0x07` endpoint in the PT
4. Register app, successful PTU should occur
5. Unregister app
6. Change the kilometers to 5000 to trigger a PTU

##### Expected Behavior
1. SDL sends a `BasicCommunication.PolicyUpdate` request to the HMI.
2. The HMI sends a `BasicCommunication.OnSystemRequest(PROPRIETARY)` notification to SDL with a valid PTS `fileName`.

##### Observed Behavior
1. SDL does NOT send a `BasicCommunication.PolicyUpdate` request to the HMI.
2. The HMI sends a `BasicCommunication.OnSystemRequest(PROPRIETARY)` notification to SDL with `fileName: null`.

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: origin/develop (https://github.com/smartdevicelink/sdl_core/commit/94f92bb7eb440c0ee878896b6281f38009d64094)
* Testing Against: Test Suite

dboltovskyi, 2020-03-11:
@ShobhitAd If you don't mind we would like to help you with creating of ATF script for this issue. This will allow us not to face with such issue in future releases since this script will be included in our regression testing scope.

ShobhitAd, 2020-03-11:
Closed via #3280

ShobhitAd, 2020-03-11:
@dboltovskyi Sorry I just saw your comment. That sounds like a good idea. Please let me know when you make a PR for the testing the issue.

ShobhitAd, 2020-03-19:
Closed via #3294

ShobhitAd, 2020-05-18:
Closed via #3363

PR #3280 (2020-03-11): Fix/SDL does not send BC.PolicyUpdate to HMI if no apps are registered

Fixes #3279 

This PR is **not ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Test Suite app and ATF

### Summary
Temporary fix: Change the `PolicyHandler::CanUpdate` function to return true even when no apps are registered

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
PR #3294 (2020-03-19): Fix/Checking PTU triggers when no apps are registered in regular policy mode
Fixes #3279 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF tests 019 and 020 in https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2368

### Summary
Currently, the IgnitionCycles and Days PTU triggers are not checked in the regular policy mode when SDL starts. In the external policy manager, this check is triggered in `PolicyManagerImpl::SetUserConsentForDevice` when `StartPTExchange` is called. `PolicyManagerImpl::SetUserConsentForDevice` seems to be mis [... cut]
PR #3363 (2020-05-18): Fix/Add Startup PTU trigger check
Fixes #3279 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF Test and manual tests

### Summary
Currently, the IgnitionCycles and Days PTU triggers are not checked in the regular policy mode when SDL starts.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-05-18 by ShobhitAd