smartdevicelink/sdl_core #3284: SDL does not set the sdlVersion in ini file

GetmanetsIrina, 2020-03-11:
### Bug Report
SDL does not set the `sdlVersion` in `ini` file

##### Reproduction Steps
1. SDL sources are in repo
2. Create a build folder
3. Go to the build folder and perform `cmake <link_to_sdl_sources>`
4. Perform `make` and `make install`
5. Check the `ini` file in `build_folder/bin`

##### Expected Behavior
sdlVersion has value of the current repo commit

##### Observed Behavior
sdlVersion is undefined

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: 94f92bb7eb440c0ee878896b6281f38009d64094

SNiukalov, 2020-03-31:
@theresalech please triage this issue. Fix is ready in [#3310](https://github.com/smartdevicelink/sdl_core/pull/3310)

theresalech, 2020-03-31:
@SNiukalov can you please advise if this is a regression issue introduced by 6.1, or if this is an existing issue in SDL Core?

SNiukalov, 2020-03-31:
@theresalech yes, this is an existing issue in SDL Core.

theresalech, 2020-03-31:
@SNiukalov thank you, understood.  We will work to include a fix in the Core 6.1 release if time permits, but will not have this included in the first release candidate which will be delivered to the Steering Committee today (2020-03-31).

iCollin, 2020-04-02:
Closed via #3310

PR #3310 (2020-04-02): Fix/Set the sdlVersion in ini file
Fixes #3284 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Manual testing

### Summary
Updated path to smartDeviceLink.ini
in CMakeLists.txt for the install command.
sdlVersion is not defined in smartDeviceLink.ini
because the file was copied from the source directory
instead the build directory.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-04-02 by iCollin