smartdevicelink/sdl_core #3285: SDL stops responding in case it has been started with EnablePolicy=false

GetmanetsIrina, 2020-03-11:
### Bug Report
SDL stops responding in case it has been started with EnablePolicy=false

##### Reproduction Steps
1. `EnablePolicy` value is set to `false` in `ini` file
2. Start SDL
3. Connect mobile device and try to register app

##### Expected Behavior
SDL connects the device and registers the app successful

##### Observed Behavior
SDL disconnects the device after the connection and stops responding

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/commit/3dcc409766b49f9914963a5c7b2ef0c0dc5fddc8
* ATF: develop
* Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/4_6/842_App_does_not_activate_when_policies_are_disabled.lua

##### Logs:
[Script_1.zip](https://github.com/smartdevicelink/sdl_core/files/4317796/Script_1.zip)

GetmanetsIrina, 2020-03-12:
@theresalech, this is the regression issue for 6.1 release. Should we prepare the fix?

theresalech, 2020-03-12:
@GetmanetsIrina yes, if you can prepare a fix, that would be much appreciated!  Please let us know once a PR is ready for review.  Thank you!

jacobkeeler, 2020-03-23:
Closing with the merge of #3292

PR #3292 (2020-03-23): Fix sdl crash when policies disabled
Fixes #3285 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF and unit testing

### Summary
When `EnablePolicy=false` inside smartdevicelink.ini file pointer to `PolicyManager` inside of `PolicyHandler` class is not set.

This way when `PushAppIdToPTUQueue` method was called there were no check for `PolicyManager` pointer validity, and so core crash happened on `policy_manager_->UpdatePTUReadyAppsCount()` call. So check for enabled policy added.

### CLA
- [X] I have signed [the CLA](https://docs.google.com [... cut]
closed 2020-03-23 by jacobkeeler