smartdevicelink/sdl_core #3295: There is no technical **sdl_core** info in README.md

MNikora, 2020-03-19:
### Bug Report
There is no info about **sdl_core** dependencies, packages, used technologies, "how to build", "how to start", etc. Also [**Getting started**](https://smartdevicelink.com/en/guides/core/getting-started/install-and-run/) link is opened not found 404 page.

JackLivio, 2020-03-19:
@MNikora The info is in the develop branch of core guides found here https://github.com/smartdevicelink/sdl_core_guides/blob/develop/docs/Getting%20Started/Install%20and%20Run/index.md

The links are currently broken but will be resolved when the core guides release is made into master. 

Please close if the issue is resolved

JackLivio, 2020-03-19:
Preview found here: https://smartdevicelink.com/en/guides/pull_request/core/62ba53a928632228d140fd135b69950dc96e775e/getting-started/install-and-run/

MNikora, 2020-03-20:
@JackLivio thank you for explanation. This ticket should be closed.

closed 2020-03-20 by MNikora