smartdevicelink/sdl_core #3302: SDL sends BC.GetSystemInfo twice

LitvinenkoIra, 2020-03-23:
### Bug Report
SDL sends BC.GetSystemInfo twice on **EXTERNAL policy** flow

##### Reproduction Steps
1. Build SDL with  **EXTERNAL policy**
2. Start HMI and SDL

##### Expected Behavior
SDL requests BC.GetSystemInfo one time

##### Observed Behavior
SDL requests BC.GetSystemInfo twice

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: develop branch https://github.com/smartdevicelink/sdl_core/commit/9118fcdf8d5bfad411444513d44a59c8dfa6ff41
* Testing Against: atf test script

##### Test Case, Sample Code, and / or Example App
TODO: Add link to the PR with atf test scripts

LitvinenkoIra, 2020-03-30:
@theresalech please triage this issue. This is not 6.1 regression issue, but anyway fix is ready in https://github.com/smartdevicelink/sdl_core/pull/3305 
Thank you!

theresalech, 2020-03-30:
Hi @LitvinenkoIra, thanks for letting us know.  I'll share with the team.  Please note that if it's imperative this be included in the 6.1 release, it will need to be raised to the Steering Committee as high priority.  Thank you!

iCollin, 2020-04-03:
Closed via #3305

PR #3305 (2020-04-03): Remove redundant GetSystemInfo sending
Fixes #3302 

This PR is **[not ready]** for review.

### Risk
This PR makes **[no]** API changes.

### Testing Plan
ATF test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2377

### Summary
SDL sends GetSystemInfo right after receiving BC.On_Ready from HMI in ApplicationManagerImpl::OnHMIStartedCooperation() - https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/application_manager/src/application_manager_impl.cc#L852 
But also there was one more redundant sending of GetSystemInfo from policy_manager. So I just re [... cut]
closed 2020-04-03 by iCollin