smartdevicelink/sdl_core #3321: SDL does not send OnAppPropertiesChange notification to HMI in case one of nicknames is removed

VjKlepikov, 2020-04-03:
### Bug Report
SDL does not send OnAppPropertiesChange notification to HMI in case one of nicknames is removed

##### Reproduction Steps
1. SDL and HMI are started
2. HMI sends BC.SetAppProperties request with application properties (without two nicknames) of the policyAppID to SDL
HMI → SDL : SetAppProperties ( properties:
{nicknames:{**"Hello Sdl1", "HelloSdl2"**} ,
policyAppID:"12345678",
enabled: true,
transportType:"WS",
hybridAppPreference: "CLOUD"})
3. HMI sends BC.SetAppProperties request with application properties (without one of nicknames) of the policyAppID to SDL
HMI → SDL : SetAppProperties ( properties:
{nicknames:{**"Hello Sdl1"**} ,
policyAppID:"12345678",
enabled: true,
authToken:"AuthToken8888",
transportType:"WS",
hybridAppPreference: "CLOUD"}

##### Expected Behavior
  1. SDL sends BC.OnAppPropertiesChange notification with appropriate app properties parameters (without one of nicknames) to HMI
SDL → HMI: OnAppPropertiesChange ({ properties:
{nicknames:{**"Hello Sdl1"**} ,
policyAppID:"12345678",
enabled: true,
transportType:"WS",
hybridAppPreference: "CLOUD"}

##### Observed Behavior
  1. SDL does not send BC.OnAppPropertiesChange notification with appropriate app properties parameters (without one of nicknames) to HMI

##### OS & Version Information
##### OS & Version Information
* sdl_hmi branch: [develop](https://github.com/smartdevicelink/sdl_hmi/tree/develop)
* commit: 74d24927cf77282c44678eba219adadbb6dcd072
* sdl_core branch: https://github.com/smartdevicelink/sdl_core/tree/6.1.0_RC

##### Test Case, Sample Code, and / or Example App
[Logs_OnAppProperties.zip](https://github.com/smartdevicelink/sdl_core/files/4426703/Logs_OnAppProperties.zip)

iCollin, 2020-04-16:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3336

iCollin, 2020-04-16:
@VjKlepikov Can you please create PR against [sdl_atf_test_scripts](https://github.com/smartdevicelink/sdl_atf_test_scripts) with the script to test this issue?
I'd like to merge that script also. Thank you!

closed 2020-04-16 by iCollin

VjKlepikov, 2020-04-17:
@iCollin Please, be informed we've created PR.  Please take look at https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2391

PR #3336 (2020-04-16): Fix/Send OnAppPropertiesChange in case one of nicknames is removed
Fixes #3321 

This PR is **not ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Use scripts from
[smartdevicelink/sdl_atf_test_scripts/onAppPropertiesChange_remove_nickname]( https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/defect/onAppPropertiesChange_remove_nickname/test_scripts/Defects/6_1/3321_OnAppPropertiesChange_remove_nickname.lua)
UTs

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)