smartdevicelink/sdl_core #3323: Deadlock during transport re-connect via USB

GetmanetsIrina, 2020-04-06:
### Bug Report
Deadlock during transport re-connect via USB with Android device

##### Occurrences
2/30

##### Reproduction Steps
1. HMI and SDL are started
2. Register 3 mobile apps via USB
3. Perform transport re-connect

##### Expected Behavior
Mobile apps reconnect successfully.

##### Observed Behavior
Deadlock is occurred.

##### OS & Version Information
* sdl_hmi branch: [develop](https://github.com/smartdevicelink/sdl_hmi/tree/develop)
* commit: 74d24927cf77282c44678eba219adadbb6dcd072
* sdl_core branch: https://github.com/smartdevicelink/sdl_core/tree/6.1.0_RC

##### Logs
[SmartDeviceLinkCore.zip](https://github.com/smartdevicelink/sdl_core/files/4437220/SmartDeviceLinkCore.zip)

GetmanetsIrina, 2020-04-06:
@theresalech , Please triage this issue.

theresalech, 2020-04-06:
Hi @GetmanetsIrina, can you please advise if Luxoft will be preparing a fix for this issue?

LitvinenkoIra, 2020-04-06:
Hi @theresalech, yes, we are in progress with the fix. Thank you

theresalech, 2020-04-06:
@LitvinenkoIra thank you!  Please let us know once the PR is ready for Livio review.

SNiukalov, 2020-04-15:
@theresalech PR is ready for Livio team review. Thank you.

ShobhitAd, 2020-04-17:
Closed via #3332

PR #3332 (2020-04-17): Fix/Deadlock during transport re-connect via USB
Fixes #3323 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Manual testing

### Summary
Sometimes we get a deadlock because we use the same lock in the same thread.
The situation is possible when calling the SendData, OnOutTransfer or AbortConnection functions.
They block the mutex at the start.
If in the PostOutTransfer function, which is called in SendData or indirectly in OnOutTransfer via PopOutMessage, we get the error LIBUSB_ERROR_NO_DEVICE,
this will call AbortConnection.
Which will cause a [... cut]
closed 2020-04-17 by ShobhitAd