smartdevicelink/sdl_core #3324: SDL doesn't unregister app with PROTOCOL_VIOLATION if it continues streaming data

dboltovskyi, 2020-04-06:
### Bug Report
SDL doesn't unregister app with PROTOCOL_VIOLATION if it continues stream data

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile device, register 2 mobile applications:
- `App_1`: `PROJECTION`
- `App_2`: `NAVIGATION`
2. Activate `App_1` and start streaming
3. Activate `App_2` and start streaming

SDL does:
   - send OnHMIStatus(FULL, STREAMABLE) notification to App_2
   - send OnHMIStatus(LIMITED, NOT_STREAMABLE) notification to App_1
   - send Navi.OnVideoDataStreaming(false) notification to HMI
   - start 1st 'StopStreamingTimeout' timeout

Once timeout is expired SDL does:
   - send Navi.StopStream(App_1) request to HMI
   - send EndService(VIDEO) request to App_1
   - start 2nd 'StopStreamingTimeout' timeout
4. `App_1` does not stop streaming, though sends EndServiceAck within 2nd 'StopStreamingTimeout' timeout

##### Expected Behavior
Once timeout is expired SDL does:
   - send OnAppInterfaceUnregistered(PROTOCOL_VIOLATION) notification to App_1
   - send BC.OnAppUnregistered(unexpectedDisconnect=false) notification to HMI

##### Observed Behavior
SDL does not send `OnAppInterfaceUnregistered` nor `BC.OnAppUnregistered`

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/6.1.0 (d8c4ac3251)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - Script: [3139_03_Stop_streaming_forcibly_app_continue_streaming](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/6_1/3139_03_Stop_streaming_forcibly_app_continue_streaming.lua)
 - Report and Logs: [2020-04-06_10-29-01.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4439243/2020-04-06_10-29-01.tar.gz)

dboltovskyi, 2020-04-06:
Please notice this is regression issue which became reproducible after merge of #3290

dboltovskyi, 2020-04-06:
@theresalech Please triage this issue

theresalech, 2020-04-06:
@dboltovskyi thank you for reporting this issue.  Can you please advise if the Luxoft team is planning on submitting a PR to fix?  If so, we will wait to review that.  Otherwise, the Livio team can work on fixing the issue.

dboltovskyi, 2020-04-06:
@theresalech Please notice currently Luxoft team aren't planning to prepare the fix.

theresalech, 2020-04-06:
@dboltovskyi understood, thanks!  Livio will work on fixing.

iCollin, 2020-04-21:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3325

closed 2020-04-21 by iCollin