smartdevicelink/sdl_core #3326: SDL sends redundant BC.UpdateAppList and BC.UpdateDeviceList requests in case BT is turned on from SDL start

GetmanetsIrina, 2020-04-08:
### Bug Report
SDL sends redundant UpdateAppList requests in case BT is turned on from SDL start

##### Reproduction Steps
1. BT module is turned on
2. HMI and SDL are started

##### Expected Behavior
SDL sends BC.UpdateDeviceList in case of app registration of unregistration.
So after system start there are no BC.UpdateDeviceList requests.
BC.UpdateDeviceList is sent only with WebEngine device(in case SDL is build with WE ON )

##### Observed Behavior
SDL sends cyclically BC.UpdateDeviceList and BC.UpdateDeviceList requests in ~ 2seconds and then in ~40 seconds.

##### OS & Version Information
* sdl_hmi branch: [develop](https://github.com/smartdevicelink/sdl_hmi/tree/develop)
* commit: 74d24927cf77282c44678eba219adadbb6dcd072
* sdl_core branch: https://github.com/smartdevicelink/sdl_core/tree/6.1.0_RC

##### Logs
[SmartDeviceLinkCore_update_devicelist_applist.zip](https://github.com/smartdevicelink/sdl_core/files/4450504/SmartDeviceLinkCore_update_devicelist_applist.zip)

AKalinich-Luxoft, 2020-04-21:
@theresalech fix for this issue was tested and ready for Livio review.
Note, this is a regression issue found by Luxoft during release testing

iCollin, 2020-04-21:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3339

PR #3339 (2020-04-21): Fix UpdateAppList/UpdateDeviceList spam
Fixes #3326 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by manual testing

### Summary
There was noticed a continuous spam of update app and device list requests from SDL to HMI even if there is no any devices connected and apps registered.
After analysis, there was found a couple of places where logic should be updated:
- In `OnHMIStartedCooperation` don't send update app list after refreshing cloud app information as this function able to trigger the same request itself when needed
- When upd [... cut]
closed 2020-04-21 by iCollin