smartdevicelink/sdl_core #3327: SDL sends response with "resultCode: GENERIC_ERROR" on GetVehicleData request in case vehicle data is not available

VjKlepikov, 2020-04-08:
**Bug Report**
SDL sends response with "resultCode: GENERIC_ERROR" on GetVehicleData request in case vehicle data is not available

**Reproduction Steps**

   1. Start SDL, HMI, connect Mobile, register new App
   2. Send GetVehicleData with speed = true
   3. Send HMI response with {"error":{"code":9,"data": {"method":"VehicleInfo.GetVehicleData"}, message:"error message" },"id":57,"jsonrpc":"2.0"}

**Expected Behavior**
   1. SDL sends to HMI VehicleInfo.GetVehicleData: "params": { "speed":true }
   2. SDL sends to mobile response of GetVehicleData with {"resultCode":"VEHICLE_DATA_NOT_AVAILABLE", "success":false, info = "error message"}

Observed Behavior
    1. SDL sends to HMI VehicleInfo.GetVehicleData: "params": { "speed":true }
    2. SDL sends to mobile response of GetVehicleData with {"resultCode":"GENERIC_ERROR","success":false}

OS & Version Information
    sdl_hmi branch: develop
    commit: 74d24927cf77282c44678eba219adadbb6dcd072
    sdl_core branch: https://github.com/smartdevicelink/sdl_core/tree/6.1.0_RC

theresalech, 2020-04-17:
@VjKlepikov please advise if this is a regression issue introduced by Core 6.1, or if it is an issue that already exists in SDL Core.  Thank you!

VjKlepikov, 2020-04-17:
@theresalech Please notice this is regression issue for 6.1 release

PR #3335 (2020-04-17): Hotfix/3327 sdl sends generic error if vehicle data not available
Fixes #3327 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF testing

### Summary
In case of GetVehicleData error response from HMI,
SDL put only part of message to event. This way
event was not processed properly and GENERIC_ERROR
was sent to Mobile App.

This commit fixes error response message structure.

### CLA
- [X] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-04-17 by JackLivio