smartdevicelink/sdl_core #3328: SDL sends two END_SERVICE requests during streaming switching between apps

dboltovskyi, 2020-04-08:
### Bug Report
SDL sends two END_SERVICE requests during streaming switching between apps

##### Reproduction Steps
1. Start SDL, HMI, connect Mobile device, register 2 mobile applications:
- `App_1`: `PROJECTION`
- `App_2`: `NAVIGATION`
2. Activate `App_1` and start streaming
3. Activate `App_2` and start streaming

SDL does:
   - send OnHMIStatus(FULL, STREAMABLE) notification to `App_2`
   - send OnHMIStatus(LIMITED, NOT_STREAMABLE) notification to `App_1`
   - send Navi.OnVideoDataStreaming(false) notification to HMI
   - start 1st 'StopStreamingTimeout' timeout

Once timeout is expired SDL does:
   - send Navi.StopStream(App_1) request to HMI
   - send EndService(VIDEO) request to `App_1`
   - start 2nd 'StopStreamingTimeout' timeout
4. `App_1` stops streaming but does not send `EndServiceAck` within 2nd `StopStreamingTimeout` timeout

##### Expected Behavior
Once timeout is expired SDL does:
   - send OnAppInterfaceUnregistered(PROTOCOL_VIOLATION) notification to `App_1`
   - send BC.OnAppUnregistered(unexpectedDisconnect=false) notification to HMI
   - not send 2nd `EndService` request to `App_1`

##### Observed Behavior
Once timeout is expired SDL does:
   - send OnAppInterfaceUnregistered(PROTOCOL_VIOLATION) notification to `App_1`
   - send BC.OnAppUnregistered(unexpectedDisconnect=false) notification to HMI
   - send 2nd `EndService` request to `App_1`

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/6.1.0 (5954b40392)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - Script: [3139_04_Stop_streaming_forcibly_app_does_not_send_EndServiceAck](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/6_1/3139_04_Stop_streaming_forcibly_app_does_not_send_EndServiceAck.lua)
 - Report and Logs: [2020-04-08_13-42-44.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4451984/2020-04-08_13-42-44.tar.gz)

dboltovskyi, 2020-04-08:
Please notice this is regression issue which became reproducible after merge of #2425

dboltovskyi, 2020-04-08:
Reproductions steps are very similar to the #3324 however the last step is different.
Also please notice current issue became reproducible due to merge of another PR.

dboltovskyi, 2020-04-08:
@theresalech Please triage this issue. Luxoft team is going to prepare a fix.

LitvinenkoIra, 2020-04-15:
@theresalech Fix is ready https://github.com/smartdevicelink/sdl_core/pull/3334

jacobkeeler, 2020-04-20:
closing with the merge of #3328

PR #3334 (2020-04-20): Do not send two END_SERVICE requests during streaming switching between apps 

Fixes #3328 

This PR is **[not ready]** for review.

### Risk
This PR makes **[no]** API changes.

### Testing Plan
[ 3139_04_Stop_streaming_forcibly_app_does_not_send_EndServiceAck](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/6_1/3139_04_Stop_streaming_forcibly_app_does_not_send_EndServiceAck.lua) 

### Summary
Check if EndService has been sent before sending it during unregistration

### Tasks Remaining:
- [x] Check regression
- [x] Check unit tests

### CLA
- [x] I h [... cut]
closed 2020-04-20 by jacobkeeler