smartdevicelink/sdl_core #3330: SetInteriorVehicleData request is processed as invalid without deprecated mandatory parameter

GetmanetsIrina, 2020-04-09:
### Bug Report
`SetInteriorVehicleData` request is processed as invalid without deprecated mandatory parameter

##### Reproduction Steps
1. HMI and SDL are started
2. Remote_control application with version 6.0 and upper is registered and activated
3. Mobile app requests `SetInteriorVehicleData` without id parameter.

##### Expected Behavior
According to API:
> <param name="id" type="SupportedSeat" mandatory="true" deprecated="true" since="6.0">
So `SetInteriorVehicleData` request should be processed successfully without deprecated parameter.

##### Observed Behavior
SDL rejects `SetInteriorVehicleData` request with `INVALID_DATA` resultCode because of deprecated parameter missing

##### OS & Version Information
* sdl_hmi branch: [develop](https://github.com/smartdevicelink/sdl_hmi/tree/develop)
* commit: 74d24927cf77282c44678eba219adadbb6dcd072
* sdl_core branch: https://github.com/smartdevicelink/sdl_core/tree/6.1.0_RC

##### Logs
[SmartDeviceLinkCore_seat_id_deprecated.zip](https://github.com/smartdevicelink/sdl_core/files/4456461/SmartDeviceLinkCore_seat_id_deprecated.zip)

GetmanetsIrina, 2020-04-09:
@theresalech, please triage this issue.
This is not the regression issue, but we propose to include this issue to Release6.1 scope,
because the requesting of `SEAT` parameters update from a mobile for new apps(starting from 6.0 version) is blocked (new apps do not use deprecated `id` parameter) by this issue.

theresalech, 2020-04-09:
@GetmanetsIrina we are currently focusing on 6.1 regression issues, but will work to include this in the second 6.1 Release Candidate if time allows.  If it's imperative this be included in the release, please raise this issue in the upcoming Steering Committee meeting as priority.  If you could please also let us know if Luxoft plans to submit a PR for this issue, with anticipated timing, we'd appreciate it.  Thanks!

GetmanetsIrina, 2020-04-10:
@theresalech, Luxoft team is going to prepare a fix.

GetmanetsIrina, 2020-04-21:
Issue is closed as not actual according to https://github.com/smartdevicelink/sdl_core/pull/3333#issuecomment-615371612

closed 2020-04-21 by GetmanetsIrina